j-kaltes/Juggluco #258: Dexcom One+ and microG

Thomas-git, 2025-01-21:
Hello,
First of all, thank you for your amazing work.

I am currently using Juggluco with Libre2 and have had great success. I am now considering switching to Dexcom One+ (since the G7 is not yet available in France).

- Is your G7 code compatible with the One+? (It seems to be the case with xDrip+.)
- Is scanning the barcode the only way to start a G7/One+ sensor? This method does not work with microG (no Play Services). I tested it on LineageOS 21 with microG and received the error: "ModuleInstall.API not available." On LineageOS 20 with microG, I got stuck on "waiting for barcode module to download." Would it be possible to use QR scanning capabilities from a third-party app like [Privacy-Friendly QR Scanner](https://github.com/SecUSo/privacy-friendly-qr-scanner), or perhaps add a simple option to input the sensor's 4-digit code manually?

Can I assist with these issues? I may be able to get a testing One+ sensor or explore a workaround for the Play Services dependency problem. While I am new to Android development, I am an experienced web developer and would be happy to contribute.

j-kaltes, 2025-01-21:
Juggluco also works with ONE+.
The version from https://www.juggluco.nl/Juggluco/download.html also works without Google Play. 
How did you get the Google Play version without Google Play?

Thomas-git, 2025-01-21:
Great news, thank you!
Using Aurora Store, I’m able to access the Play Store app. It’s quite useful for receiving automatic updates when apps aren’t available on F-Droid.

I also run a web server—would it be helpful if I set up an F-Droid repository for Juggluco? Or would that be too much overhead for you to manage?

j-kaltes, 2025-01-21:
It is already at Uptodown: https://juggluco.en.uptodown.com/android
That that also contains the ZXing barcode library.