j-kaltes/Juggluco #264: Troubleshooting Libre3 Sensor Connection Issues When Running the Latest Code

Abdullah6445, 2025-02-06:
I'm encountering a problem connecting to my Libre3 sensor.  While the official Libre3 app from the Play Store connects and retrieves readings without issue, I'm experiencing difficulties when running the latest code (presumably code you provided).  I'm receiving an "Unrecognized NFC scan error."

This error indicates a problem with the NFC communication between the code and the Libre3 sensor.  I'm seeking assistance in resolving this "Unrecognized NFC scan error" so I can successfully connect to my Libre3 sensor using the provided code. What are the potential causes of this error in this specific context, and what debugging steps should I take to identify and fix the problem?

Abdullah6445, 2025-02-10:
@j-kaltes Thanks for your help. I have a few more questions. I've tried turning NFC off and on, and also rebooting my phone, but the app is still crashing. The app from the Play Store works perfectly fine on this phone, but the app I built myself from the code crashes on the same device. I don't think it's an NFC issue.  To help me troubleshoot, could you please tell me:
What is the active build variant? (e.g., mobileLibre3SiDexGoogleRelease )
What is the active ABI? (e.g., armeabi-v7a, arm64-v8a, x86)

j-kaltes, 2025-02-10:
The issue seems to have changed. I understood that you first said you could scan the sensor with Abbott’s Libre 3 app from Google Play, but not with Juggluco from Google Play. Now you seem to say that you can use the sensor with Juggluco from Google Play, but that the Juggluco you build yourself does crash.

Abdullah6445, 2025-02-10:
I said that when I install the Juggluco app from the Play Store and enter my credentials, I can scan the sensor there. But when I run your code, I am unable to scan the sensor, and the app crashes.

j-kaltes, 2025-02-10:
When you said "_While the official Libre3 app from the Play Store connects and retrieves readings without issue_", you were talking about Juggluco?

Abdullah6445, 2025-02-10:
yes

Abdullah6445, 2025-02-10:
@j-kaltes  I'm building the Juggluco app from the Juggluco repository code to connect to the Libre3 sensor and retrieve readings. I enter my credentials in the "Exchange Data" section and retrieve the account ID from LibreView. When I try to scan the sensor to establish a connection, my app crashes. However, when I download the Juggluco app from the Play Store and follow the same process, I encounter no issues. The sensor connects and provides readings. Why is that? i hope this clears everything?

Abdullah6445, 2025-02-10:
@j-kaltes ??