j-kaltes/Juggluco #270: Question: Smartphone <=> smartwatch communication

defaultuser0815, 2025-02-15:
Hi, I'm confused how to establish a "consistent configuration" wit Juggluco (Freestyle Libre 3 EU) for my smartphone (Android 14) and smartwatch (Galaxy watch 4, One UI 6).

What is most important: I can receive glucose values with the smartphone as well as with the smartwatch. This is awesome and I am very thankful for that. 

Yet, I have some questions / suggestions.
I would like to have my smartphone as a "primary" collector for the sensor. If the smartphone gets out of range it would be great if connection smartwatch - sensor could be established automatically. How can I configure this?

At the moment I have errors "not receiving data from clone" on both device not receiving data from the sensor. Having smartphone AND smartwatch as a collector from time to time, what is the configuration for the "Master" - Clone settings? 

Can you help me here?

j-kaltes, 2025-02-15:
You have to switch between phone and watch yourself with left menu->Watch->WearOS config->_Direct sensor watch connection_.
The phone doesn't know that is has a disconnection error because you left the house or something else.

defaultuser0815, 2025-02-15:
Thanks.
My default setting is: "Use Bluetooth"  is checked / activated on my smartphone and unchecked / deactivated on my watch
"direct sensor watch connection" is activated on my smartphone. 

I leave my home without the smartphone. On my watch I check "use Bluetooth" and I have the glucose values directly on my watch. Awesome! 

I come back home. Smartphone gets no values from the watch (no data from clone). 
- What do I have to do to get the smartphone back as a collector? Ideally, I would like to operate on my smartphone only (e.g. re-activate "use Bluetooth"?). Is this possible?
- What do I have to do so that the values received with the watch before are transferred to the smartphone?

j-kaltes, 2025-02-15:
You shouldn't use "use Bluetooth" directly, but, as I said already, check left menu->Watch->WearOS config->Direct sensor watch connection. Otherwise you also have to change the mirror settings and change "Use Bluetooth" on the other side of the connection.

defaultuser0815, 2025-02-15:
Sorry, if I wasn't clear: 
left menu->Watch->WearOS config->Direct sensor watch connection. I _have activated_ this. 

Can you confirm that in case I use the watch as a collector and want to switch to have the smartphone as a collector I have to activate "use Bluetooth" on my phone and _and also deactivate_ "use Bluetooth" on the smartwatch?

In this case my feature request would be to have a switch on both devices: Activating "Use Bluetooth" in one device should deactivate "Use Bluetooth" on the other device. That would be fantastic for my purposes :)

j-kaltes, 2025-02-15:
No you have simply to uncheck left menu->Watch->WearOS config->Direct sensor watch connection, when you want the connect the phone directly with the sensor.

defaultuser0815, 2025-02-16:
Tested it: 
T_0: Phone as a collector. "Direct sensor watch collection" unchecked.
T_1: Phone switched off. "use Bluetooth" Checked on Smartwatch. (Went for a run.)
T_2: (Came back.) Phone switched on. "Use Bluetooth" checked on phone. Are a couple of minutes connection with the sensor seemed to come back.
So, this works. Thank you! 

What is unclear to me is the "synchronisation" of the glucose values. My impression is that when I get back to my smartphone that glucose data is backfilled, _fetched from_ _sensor_ but it is _not downloaded from the watch_. 
Is there a way to make this possible? Change some settings related to clone?

j-kaltes, 2025-02-16:
Do you mean that a period of disconnection, your watch received past glucose values, but they are not send to the phone? But the current values are present on the phone?
Can you give some screenshots illustrating this?

defaultuser0815, 2025-02-17:
I'm not sure whether screenshots can illustrate it properly. 
I think, when I receive values with my smartphone, they are transferred to my watch via Bluetooth. 
If my watch is turned off for some time and I switch it back on after e.g. 2 hours, ALL values within these last 2 hours are transferred to my watch. 
The other way round: My phone is turned off and I receive the values with my watch. After switching the phone on again the values received with the watch are NOT transferred to the phone. As soon as the phone has connection with the sensor again, I receive the glucose values from the past. I think this is due to backfilling - the SENSOR delivers data off the past couple of hours. This is the Dexcom behavior.

I would prefer having a two ways data exchange between the smartphone and the watch when it comes to this.

[image]

[image]

j-kaltes, 2025-02-17:
This looks like, no values send from watch to phone because of a connection problem between phone and watch. This has nothing to do with back fill. If the mirror configuration is right (as when left menu->Watch->WearOS config->Direct sensor-watch connection is set and nothing is changed), you need to do things like press Mirror->Sync, Reinit or turn Wifi or Bluetooth off and on in the watch or phone.

defaultuser0815, 2025-02-20:
I fiddled around a little bit and it does work now, (I think 😊). Thank you.