j-kaltes/Juggluco #275: How do we diagnose "sensor error"

ks2018ks, 2025-03-03:
Hello I have Dexcom One+ working fine for 5 days and just now Juggluco says "sensor error". Is there a way to find out what kind of error? 

Could it be the sensor is 
- dislodged from the blood supply? 
- or out of battery? or 
- broken?
- or?

j-kaltes, 2025-03-03:
While using the three Dexcom G7 sensors and one Dexcom ONE+ sensors I have used, they never gave a sensor error, except when the wear duration was up, that is after 10 and a half day.
So I don’t know what is the matter. I consider it very unlikely that the battery is empty given that I can get data out of the sensor long after their wear duration.
Can you give some more information:
- A screenshot of left menu→Sensor;
- A screenshot of left menu→Sensor→Info
- Even better, also run a logging version of Juggluco from https://www.juggluco.nl/Juggluco/download.html#bugreport and turn on Logging of trace.log and logcat in Left menu->Settings-> Logging and after using the sensor for some time (30 minutes), save  trace.log and logcat.txt and e-mail them to me.

j-kaltes, 2025-03-03:
From the logs you send me, I see that the sensor somehow stopped working and is constantly sending an invalid measurement from  Mon Mar  3 09:12:09 2025. It behaves like a sensor behaves that was activated more than 10.5 days ago, only your sensor was started Tue Feb 25 14:38:23 2025, which is only 6 days ago.
So not only Abbott’s sensors can have these problems.