j-kaltes/Juggluco #279: Motorola/android 15 Bluetooth connection failure

alexmbrennan, 2025-03-07:
I have recently bought a new Motorola 50 Edge Pro currently running Android 15 to replace my old Samsung S10E and I have not been able to establish a Bluetooth connection to the sensor at all with the new phone.

I have switched off the old phone, I have turned off Bluetooth privileges for the LibreLink app and the sensor is presumably fine since it works with Juggluco on the old phone. NFC scans work fine, but Bluetooth does seem to work at all. I am seeing status=147 a lot. I have tried every combination of "Turbo" and "Android" Bluetooth settings without success. 

I also believe that the same issue was present on Android 14 but since the phone updated to Android 15 on the day I bought it I am not able to investigate this further. 

I will try to email you the log files once I can.

j-kaltes, 2025-03-07:
Strange immediately after the establishment of a connection it is disconnected. 
Does the sensor still work without problems with the old phone?
Did you try rebooting the phone?
Did you try doing a network reset?
Can you return the phone to the seller?

alexmbrennan, 2025-03-07:
The sensor seems to be working fine with the old phone (just the expected connection issues when I put the phone in the wrong pocket, etc).

I have rebooted the new phone several times, and I just reset the Bluetooth and Wifi settings and rebooted the phone again just to make sure but juggluco still isn't able to connect to the sensor.

I don't think that I can return the phone for a refund without showing that it's defective because I unfortunately bought it in person. 

I guess I will try a factory reset after I get a replacement SIM before I dispose of the Motorola but I don't expect much since it's brand new.

alexmbrennan, 2025-03-08:
I just did a full device reset and the device still has the same issue so I am going to returning it (or selling it on eBay or whatever).

Thank you for your insight but if it works on Android 15 and Motorola phones in general then it has to something wrong with this particular device so there is probably not much point in pursuing this further.