j-kaltes/Juggluco #282: Juggluco vs LibreView data smoothing

jrdeutsch, 2025-03-14:
Hi - seeing quite different data chart results between the two tools.  Wondering if there's a way to "smooth" out Juggluco so that it compensates better for the sensor raw data wackiness?

[image]
[image]

j-kaltes, 2025-03-14:
I see exactly the same value 84 mg/dL in LibrelinkUp and Juggluco and that is always the case. The curve in LibrelinkUp are not the stream values, but the history values in Juggluco.

jrdeutsch, 2025-03-14:
Technically, I understand what you're saying. I can see how the history lines match LibreView's lines.  However, what I still don't understand is how the "stream" lines are so wildly drastic and _seemingly_ inaccurate.  

For instance, you can see that the huge dips and spikes in the Stream report jump wildly between 4:30AM and 5:00AM well beyond a 5-minute interval....

j-kaltes, 2025-03-14:
The jumps up and down of the stream values has nothing to do with Libreview.
The history values jump less up and down, but are at least 20 minutes behind.

jrdeutsch, 2025-03-14:
Right. I'm not talking about Libreview anymore...just Juggluco's charting.
Any idea why the "stream" lines are bouncing all around seemingly
inaccurately? Is the sensor really telling Juggluco every minute that my
glucose is changing 80 points suddenly?

j-kaltes, 2025-03-14:
Libre 3 values come directly from the sensor. The sensor gives every minute a value, it doesn't relates it to previous values. The measurements are far from perfect, but maybe the blood glucose level really goes up and down.

jrdeutsch, 2025-03-14:
This is quite unlikely, which is why I wonder why there's such a difference
between Juglucco's reporting and the Abbott Libre app, which appears to
smooth out the readings (even though it also updates every minute).  Just
interesting; wondering if the Abbott app actually adds a smoothing
algorithm whereas Juggluco does raw data reporting.

j-kaltes, 2025-03-15:
As I said, the history values in Juggluco are the same as the curve values in Abbott. If you didn't turn of the display of history (https://github.com/j-kaltes/Juggluco/discussions/268), there were no such differences.