j-kaltes/Juggluco #283: Sensor-Watch connection won't get past handshake

tofof, 2025-03-16:
I'm trying to use Juggluco so that the sensor is synched to my watch, instead of to my phone.

I activated the FSL3 with juggluco on the phone and used just it for the first several days, and now have installed juggluco onto the watch (and gave permissions) and toggled 'Direct sensor-watch connection' as instructed. As expected, the phone app switched 'use bluetooh' off, while the watch switched it on. 

However, the watch never receives data from the sensor.

I have tried:

- rebooting the watch
- rebooting the phone
- toggling the connection
- reinitializing the watch connection
- resetting defaults for watch connection
- using with or without turbo
- using with or without android
- toggling bluetooth on watch
- disabling wifi on watch
- disabling bluetooth on phone
- powering phone off
- moving the watch within 10 cm of the sensor for 20+ minutes
- uninstalling juggalo completely on watch, then reinstall (including deleting stale mirror connection on phone) and reenable permissions

[image] [image]  [image]

It never gets farther than here. These screenshots are at 19:01, so 42 minutes after the handshake. The phone had bluetooth off or was completely powered down the entire time, up until I rebooted it to take screenshots (this watch can only do that through the phone). 

The data in 'info' is always the same - the hex string `0898`...`cce4`, unless I view it very quickly after toggling to the watch. In those cases it will briefly show `SEC_CHAR_CHALLENGE_DATA`, presumably part of the initial handshaking. The 'failure' line that ends up later with strikethrough about `status=133` shows up about half of my attempts. 

Here is another example, this time trying Android checkbox (I know it is discouraged and it was one of the last things I attempted before filing this issue). If it matters, I have developer mode on and adb debugging enabled. 

[image] [image] [image]

Hardware: OnePlus Watch 2 (OPWWE231), Google Pixel 4a(5g)
Watch OS: WearOS 4 + OxygenOS version 6 (RTOS)
Juggluco
&nbsp;&nbsp;&nbsp;&nbsp;Phone:  9.1.2-phone-li3-si-x-g 649
&nbsp;&nbsp;&nbsp;&nbsp;Watch:  9.1.2-wear-li3-si-x-g 650

tofof, 2025-03-16:
[EDIT - split top post to just issue, moving troubleshooting steps to replies]

I have tried:

- rebooting the watch
- rebooting the phone
- toggling the connection
- reinitializing the watch connection
- resetting defaults for watch connection
- using with or without turbo
- using with or without android
- toggling bluetooth on watch
- disabling wifi on watch
- disabling bluetooth on phone
- powering phone off
- moving the watch within 1-10 cm of the sensor for 20+ minutes
- enabling the juggluco watchface
- leaving the watch on charger (and dev option of screen on while charging), and juggalo app graph in foreground for 20+ minutes
- enabling AOD and leaving watch on juggluco watchface for 20+ minutes, off charger
- toggling back to phone instead of watch-sensor direct, getting data, toggling back to watch-sensor
- uninstalling juggalo completely on watch  (including deleting stale mirror connection on phone), then reinstall and reenable permissions
- several periods of 60+ minutes total for the watch to be responsible for the sensor, with watch and sensor within 40 cm

j-kaltes, 2025-03-16:
You say that you have tried it for 20 minutes, but on the screenshots I see that you turned on "_Use Bluetooth_" less than one minute before making the screenshot.
You could install a logging version of Juggluco over the current one from
https://www.juggluco.nl/Juggluco/download.html#bugreport
Let it try to get a connection for 20 minutes without interrupting it, while you have turned off Bluetooth in the phone and e-mail send trace.log to me.

tofof, 2025-03-16:
I apologize if I mixed up which screenshots were exactly at which moments. I assure you I have taken all the steps I claimed. 

[watch-trace.log](https://github.com/user-attachments/files/19273073/watch-trace.log)
[phone-trace.log](https://github.com/user-attachments/files/19273074/phone-trace.log)
[phone-logcat.txt](https://github.com/user-attachments/files/19273075/phone-logcat.txt)

Here are logs after 20+ minutes. I left the watch sitting with juggluco in foreground the entire time and changed nothing on the watch in that time. After enabling watch-sensor connection and taking an initial screenshot, I placed the phone into airplane mode and waited until the 20 minute no data alarm went off, just as before. 

Here are screenshots at the beginning and end of this wait. As you can see, the handshake line is the same across the entire 20 minute period (I watched throughout), the `08981482-ef8911e9-81b4-2a2ae2dbcce4` entry. 

[image]
[image]
[image]
[image]

And here is the watch's log after I've written this up, 40+ minutes. 

[trace.log](https://github.com/user-attachments/files/19273143/trace.log)

j-kaltes, 2025-03-16:
Everything seems to work, except that it doesn't receive glucose values. Does the sensor still work with the phone?
You seem to have done everything you can to make it work.
Do other sensors work with that watch?

tofof, 2025-03-16:
Yes, the sensor still works fine with the phone whenever I switch back. 

I don't know, this is my first sensor; I suppose in another 7 days I can try the next sensor with the watch. I know the bluetooth on the watch is at least functional; it paired with my phone of course and it can see the devices I expect when it's in pairing mode. 

I had purchased the watch specifically for this, so it's frustrating that it didn't work. 

In the meantime today I have gotten an alternative watch, a Samsung 6 Classic, which took over the sensor easily. So, as far as solving my problem, I can keep that one instead.  Even so, defective hardware seems like such an unlikely cause. I will at least wait a week and try with the next sensor out of my own curiosity -- Is there anything other than using the logging versions that I should do with it?

j-kaltes, 2025-03-17:
With "_other sensors_" I meant other sensor types, like Libre2. I assumed that all Libre 3 sensors would have the same problem.