j-kaltes/Juggluco #285: Able to scan Libre 2 via NFC but unable to connect via Bluetooth

dcaguiat, 2025-03-19:
I've managed to install Juggluco on my son's android and we are able to get a reading from FSL2 by NFC scanning. There seems be a problem connecting to Bluetooth though, it says 'Use Bluetooth' off where the reading should show but as far as I know, Bluetooth is on, and I can't find any switch to toggle the Bluetooth connection.

[image]

j-kaltes, 2025-03-19:
"Use Bluetooth" is not Bluetooth. You can turn it on in left menu->Sensor. If you didn't turn it off yourself, it is probably turned off because Juggluco lacks certain permissions needed for Bluetooth, for example _Nearby device_ permission on Android 12 and higher.

dcaguiat, 2025-03-19:
That's the setting I was looking for but couldn't find. Thanks a lot, it works now! :-)

closed 2025-03-19 by dcaguiat