j-kaltes/Juggluco #287: 2 Libre 3 Pllus sensors at same time

source2k, 2025-03-25:
I remember that it was possible to have 2 Libre 3 sensors running at the same time to compare the values.
Currently with 2 Libre 3 Plus sensors only values for the new sensor are displayed. The old sensors has the state "no connection".

Is it possible to have 2 Plus sensors at the same time?

j-kaltes, 2025-03-25:
If the device address in left menu->sensor is yellow you can, if it is red you can not.
You need to start Juggluco first and when it is receiving values start Librelink.
You can also compare values by scanning the sensor.

source2k, 2025-03-25:
The device addresses for both sensors are yellow and parallel reception does not work.
The glucose value for the new sensor is displayed.
No connection is displayed for the old sensor.
I scanned the old sensor and no value is received.

I suspect that Abbott "optimized" the new better Plus sensors to prevent comparison.

j-kaltes, 2025-03-25:
Your exact situation is unclear to me:
Which sensor is the current sensor of Abbott's app?
Where are values not displayed?
Scanning the old sensor with what? How are values not received? Via Bluetooth?

In any case you can simply compare values by scanning the sensor.

source2k, 2025-03-25:
I'm not using the Abbott app.
I started the sensors with Juggluco and I'm using only Bluetooth.
The start of the old sensor was 14 days ago. I started the new sensor today, because I wanted to compare the new and the old sensor chart.
Maybe I remember wrong, but I think this was possible.

Now, on the right side of Juggluco, the new sensor and value is display at the top.
The old sensor at the bottom shows "no value". The display of the value stopped at the start time of the new sensor.

A NFC scan of the old sensor does not change anything.

j-kaltes, 2025-03-25:
I totally misunderstood your question.

Normally you can run two sensors at the same time.
To get a sensor working, you need to do the same things as when you have only one sensor.
I heard of users of Libre 3 sensors who were not able to run two sensors at the same time on one phone, but could do so on another phone. I have never experienced any difficulty running two sensors in parallel, if anything it works better than running one sensor on it's own, especially on WearOS watches. Probably because it keeps the device awake.

source2k, 2025-03-25:
OK, I'll try it again at the next sensor change. Maybe its just the phone.
Thank you for you great work on Juggluco. It would drive me crazy, if I had to use Abbotts app without the possibility of having the glucose value on my watch.

source2k, 2025-08-03:
Now I had to replace a still active Libre 3 Plus sensor by a new one because of bad measurements.

I think I found reason for Status=133 on my phone:

1. If several Libre 3 sensors are in range, a mobile phone may only connect to one and mostly the same (maybe because of internal driver timings?).
2. A senor connection may be lost in favor for the connection to another sensor.
3. If no sensor is connected (Status=133) in juggluco at all, you can try the app "nRF Connect" and wait for the message "XX:XX.. has connected" (XX:XX is the MAC address), to find out which sensor wants to connect
4. The MAC address of the last two sensors are found in juggluco sensor menu (use drop down list to select another sensor).
5. The solution for me was to get  out of the Bluetooth range of the undesired sensors (you can also put them into a closed metal pot) and restart the mobile phone.

uschindler, 2025-11-08:
Please change issue title to "Two Libre 3 / 3+ sensors at same time broken on some phones".