j-kaltes/Juggluco #289: WearOS Configuration Question

dtregobov, 2025-03-31:
Long story short, currently using direct sensor connection on WearOS watch with LTE so i can put a little smart watch on my cats collar to monitor him. The issue im facing is, currently the watch mirrors the info to the phone, then the phone exchanges to Libre view. This works but when my cat wanders out of bluetooth range it obviously stops. Can you send values directly from WearOS to libreview? If not, what is a good alternative? I am unsure what apps or server i could use to allow the watch to upload the readings and then be viewed remotely.

Thanks in advance.

j-kaltes, 2025-04-01:
If you aim is to follow the glucose values over the internet, an alternative is to use left menu->Mirror.
You can either keep an Android phone running Juggluco at home and forward a port from your modem to that phone (https://www.juggluco.nl/Juggluco/phones/index.html) or you can run Juggluco server on a computing cloud (https://www.juggluco.nl/Juggluco/cmdline/watch).

In all cases you should not use the automatically created mirror connection between the phone and the watch for this purpose, but create a new connection. You can either use a different phone or maybe you can turn off left menu->Watch->WearOS on the phone and turn off or delete the automatically created mirror connection.

Is your cat ill or guinea pig?

dtregobov, 2025-04-01:
Awesome, thanks a ton. I will try these suggestions, but I may be back with more questions because this stuff is a little bit outside my skill set. Unfortunately, my cat developed diabetes as a side effect of taking steroids for a separate issue. Luckily we are managing it pretty well and he is doing okay, but he hates being cooped up in the house so I am on the hunt for a solution. Luckily juggluco has worked awesome on a little galaxy watch that I put on the collar, after this last step we will be able to let him go off on his adventures and track him from home.