j-kaltes/Juggluco #290: Fail to connect to Sibionics (CN version)

1nn0, 2025-04-06:
Hello!
I recently replaced smartphones for my kids. Old ones was in FUBAR state.
New ones are Android 15 Motorola Edge 50 Neo. 
While using Libre 2 there are no problems so far.
Earlier today I tried to connect to Sibionics sensor. QR-code scanned without issues, I see sensor ID, but the address is always unknown. 
I tried to connect this sensor to my Google Pixel 7 and connection was successful (I see sensor ID, BT address and glucose data)
Installed a log version of juggluco on one of my kids phone and tried to connect from scratch.
What I tried:
1) Clear cache and data of the app - no luck
2) Reset BT and network settings - no luck
3) Rollback to Juggluco 9.1.3 (this version was on old phones) - no luck
4) Use Turbo and Android options in various combinations - no luck

Logs are in attachment. Any help will be greatly appreciated.

Phone: Moto Edge 50 Neo
OS: Android 15
Juggluco version: 9.3.2-log (google drive version)
Trace Log: [trace.log](https://github.com/user-attachments/files/19621367/trace.log)

1nn0, 2025-04-06:
Fresh logs (~45 minutes). 
Rebooted before and during logging.

[trace.log](https://github.com/user-attachments/files/19621853/trace.log)

[logcat.txt](https://github.com/user-attachments/files/19621856/logcat.txt)

j-kaltes, 2025-04-06:
Again, scanning for Bluetooth devices returns no devices. 
Did you try https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan ?

1nn0, 2025-04-06:
Follow up:
Something strange happened.
First I installed BLE Scanner on my phone (pixel 7) to check if the sensors were available and visible. They were.
Next I installed BLE Scanner on phone of one of my kids. Started it, gave permissions. In the moment a saw sensor (in BLE scanner App) it connected to Juggluco (notification popped with new glucose value). Also it got all history data from sensor (like 3 hours of measurements).
Ok, weird, but ok.
I installed same scanner to another kid's phone. Launched it. The moment I see sensor (different one) - Juggluco connects to it. History data, notifications, NS-exporter - all good.
I asked my kids to go to another room without their phones to test if there are any issues with reconnecting to sensor (when connection is lost, like of they forget to take phones any way they go). Reconnected perfectly fine within 2 minutes.

I think, that it is some rare compatibility issue, mostly from phone side.
I can provide additional logs and any other data if needed.

For now I'm fine with this kinda "workaround". 

Thank you for your reply and hint. Also, as a parent I would like to thank you so much for your hard work on Juggluco. We've been using it for almost a year now, its is rock solid and very lightweight. Perfect App for us, compared mostly against xDrip (as most close competitor). 

Anyway, I think the issue is resolved.

1nn0, 2025-04-06:
Yes, after reboot I need to launch scanner, just lauch and wait for couple of seconds.

1nn0, 2025-04-06:
> I modified Juggluco so that it now and then searches for Bluetooth devices in general, instead of filtering on Sibionics devices. Does that make a difference: https://drive.google.com/file/d/1UWYyssyhcT518gI-2spQ6hNZKducZcPi/view

I installed this version. Rebooted the phone. 
No issues with sensor, reconnected right after home screen loaded.
I went even further - stopped and deleted sensor from juggluco, cleared cache and data. Relauched app and added sibionics sensor from scratch. It worked flawlessly. 10 seconds from QR-code to successful coneсtion.

SimonPhilpott, 2025-10-25:
> I modified Juggluco so that it now and then searches for Bluetooth devices in general, instead of filtering on Sibionics devices. Does that make a difference: https://drive.google.com/file/d/1UWYyssyhcT518gI-2spQ6hNZKducZcPi/view

Is this version still available? Or did this change you made get merged.