j-kaltes/Juggluco #291: Issue with V3 Api

kenkit, 2025-04-06:
Hi, this is a great app thank you for making it.
I use it with my nightscout server I setup a few days ago.
I started using it today with Libre 3 but I don't know what happened since it stopped synching. After looking through the logs and script on my nightscout server I saw this ?
I'm not sure if the problem was caused by me trying to use the watch as an uploader to nightscout since I saw it still uses the v1 api.

I am now disabling v3 to see if the error will go away.
 [image]

kenkit, 2025-04-06:
Disabling the v3 checkbox somehow fixes the issue.

j-kaltes, 2025-04-07:
The nightscout server log will always display this Identifier mismatch message. Juggluco generated identifiers differently, but that doesn't prevent data from ending up at the nightscout server.

kenkit, 2025-04-09:
The data was not showing up on the website or apps that use nightscout, so I think there is a problem somewhere.

kenkit, 2025-04-09:
check the id of the previous messge slightly visible in the image, it's the same as the current id.

kenkit, 2025-04-09:
I will try it again and see if it actully works