j-kaltes/Juggluco #300: Units in LibreLinkUp don't match the chosen unit type in app

ItsVovix, 2025-05-01:
Hello,
I'm currently using Juggloco with a Libre 2 sensor and I'm using the Libreview uploader to share my data with others (at least at the moment before I migrate everything to Nightscout).

However, one issue that I've noticed it that the displayed glucose in LibreLinkUp doesn't match the selected one in Juggloco settings. **After I scan a sensor in Juggloco, LibreLinkUp switches the units to mg/dL regardless of the Juggloco setting.** LibreView is unaffected and shows everything as intended.

It's a minor issue so I don't think it should be given much priority, but I would appreciate it nonetheless.
_Thank you for creating and maintaining this amazing app by the way, it makes my day to day life so much easier!!_

j-kaltes, 2025-05-02:
LibrelinkUP is an app totally independent of Juggluco. If you have issues with LibrelinkUp contact the makers of LibrelinkUp.

ItsVovix, 2025-05-02:
Yes, that was my first though too. However after looking a bit into it, I'm led to believe this issue is related to Juggluco.

LibreLinkUp doesn't have a setting to change the unit type between mmol/L and mg/dL. Sensor scanned with Juggluco shows mg/dL, meanwhile when using the official (and feature lacking) LibreLink app to scan the sensor the units switch to the correct one.

I've linked a crudely edited video below to demonstrate exactly what I mean (note for replication purposes: it's needed to clear all data of LibreLink for it change the unit type)
https://streamable.com/e/5gc2si

As you can see from the video: The app first shows mg/dL after I've been using it with Juggluco, then after scanning with official LibreLink to scan, it shows correct mmol/L and then, finally after scanning with Juggluco it switches back to mg/dL with no way to change it back to mmol/L.

Thank you for your time, I hope I made it clearer now.

j-kaltes, 2025-05-02:
I also use Juggluco in mmol/L units. LibrelinkUp, Librelink and Libreview are all stuck in mmol/L. 
My country is mmol/L what country are you from?

ItsVovix, 2025-05-02:
> What country are you from?

I'm from Czechia, so my default is mmol too.
> LibrelinkUp, Librelink and Libreview are all stuck in mmol/L.

That's really interesting! Out of these theree, only LibreLink and Libre have set themselves as mmol for me.

Could it be related to the fact that I've started the sensor using the official app instead of Juggluco? Or do you have any other ideas?

j-kaltes, 2025-05-02:
Your are not using the Android version of LibrelinkUp. Android LibrelinkUp doesn't have a Darkmode .

ItsVovix, 2025-05-02:
I am though. I'm using the lastest (version 4.14.0) off the playstore [this one to be specific](https://play.google.com/store/apps/details?id=org.nativescript.LibreLinkUp). It even has a dark-mode screenshot included in the playstore app preview (or whatever it's called)
<details>
<summary> Playstore screenshot</summary>

[image]
</details>

j-kaltes, 2025-05-02:
You are right. I have on two phone installed this same version. On one it is displayed in the old fashioned white, in the other it is in black mode.

ItsVovix, 2025-05-02:
I had it as mg/dL when I first set it up, yes. Then I changed it and tried removing & re-entering my LibreView login information in hopes that it would update, which didn't work.

What fixed it was clearing all Juggluco app data and changing the unit to mmol/L **before** logging in with LibreView for the first time, not charging it after logging in.

ItsVovix, 2025-05-02:
Fixed! Need to change the unit type before the first ever LibreView sync, not after. Thanks a lot for the help!

closed 2025-05-02 by ItsVovix