j-kaltes/Juggluco #305: FR: Add export Juggluco settings to file

Stan-Di, 2025-06-04:
@j-kaltes Please add to Juggluco the ability to export settings to a file. 

Recently, a lot of settings have accumulated (web server, mirrors, export to libreview) and it is difficult to re-enter everything when changing a phone. This would make the task easier. 

And if the sensor key would present in the export file, that would be great! :)

nightscout-ricsi, 2025-06-06:
I am in the same situation. I have ggogled for juggloco export settings, and find this.

Stan-Di, 2025-08-01:
With versions after 9.7.7, things got even worse: after the update, the mirror stopped working for some unknown reason. It seemed to me that this happened when I turned on the battery saving mode, but 2 days after that, the mirror still didn't work. I rebooted the phone, deleted and re-entered the mirror settings, but it still didn't work.

@j-kaltes once again I ask you to export the settings to a file

Stan-Di, 2025-08-03:
I use Juggluco server on my host server and keep free mirrors for everyone (there are 22 of us now). The server receives data from the "master" and transmits it to the "followers". The test server is located at 99ns.ru port 8000 password 1234. For the master label **master**, for the receivers **follow** (and follow2, follow3 etc). The web page for each mirror is at jug**port**.99ns.ru for the test https://jug8000.99ns.ru/sgv.json?count=10

But in any case, there are now a lot of settings besides the mirror: alarms, Nightscout, uploading to AAPS. And considering that sometimes you have to reset the application, you have to change everything again.