j-kaltes/Juggluco #306: Not working SSL port in Juggluco server

Stan-Di, 2025-06-07:
Please help me solve the problem of Juggluco server 9.7.1-2. The server is installed on Ubuntu Arm8, all libraries are installed with libssl-dev and libgles2-mesa-dev (I don't understand why they are needed, but without them it doesn't compile). The server is built from sources and runs. But the SSL port does not open. I am attaching a screenshot with the ss command. And the http port is open and working. How can I solve this problem?

[image]

j-kaltes, 2025-06-07:
Did you use certbot to get an ssl certificate as described at https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html
and put prevkey.pem and fullchain.pem in the directory 8111?

Stan-Di, 2025-06-07:
Yes, prevkey.pem and fullchain.pem in the directory 8111, but created with 

openssl genrsa > privkey.pem
openssl req -new -x509 -key privkey.pem > fullchain.pem

j-kaltes, 2025-06-07:
Also when I create the SSL certificate that way, does Juggluco listen on port 18111.

Stan-Di, 2025-06-07:
Thanks a lot! There are problem in my system. My libcrypro in /usr/lib/aarch64-linux-gnu/, not in /usr/lib/

I use ln -s to repair it!

ln -s /usr/lib/aarch64-linux-gnu/libcrypto.so.3 /usr/lib/libcrypto.so.3
ln -s /usr/lib/aarch64-linux-gnu/libcrypto.so /usr/lib/libcrypto.so
ln -s /usr/lib/aarch64-linux-gnu/libssl.so.3 /usr/lib/libssl.so.3
ln -s /usr/lib/aarch64-linux-gnu/libssl.so /usr/lib/libssl.so

Thank for help with debug!

closed 2025-06-07 by Stan-Di