j-kaltes/Juggluco #309: Android 15 Glucose Curve Main Screen Blank Issue

birdfly, 2025-06-18:
When I swipe within the curve view, the value display area that is normally fixed on the right side of the screen does not show any glucose readings—sometimes only an arrow appears. As I continue swiping left, the originally fixed value display area shifts infinitely forward and becomes a blank screen (I understand this might be the coordinate space for future glucose values, but this behavior should not occur, as the area is supposed to show the most recent glucose readings and should not point to the future).

This issue occurs on my Android 15 device. I believe it started after upgrading to Android 15. At that time, I was using Juggluco version 9.6.1. I’m currently on version 9.7.5, and the issue still persists.

I’m using Libre 3+.

[image]

birdfly, 2025-06-18:
Device ： Oneplus 8T
System：RisingOS v6.3 Android15

let me try to get logs

j-kaltes, 2025-06-18:
https://xdaforums.com/t/rom-15-violet-unofficial-risingos-revived-v6-3-stable.4725854/ ?

birdfly, 2025-06-18:
https://xdaforums.com/t/op8t-kebab-a15-rom-watchdogs-june-6-2025.4671238/post-90027484

birdfly, 2025-06-18:
> adb shell ls -l /system/fonts

-rw-r--r-- 1 root root     4540 2009-01-01 08:00 AndroidClock.ttf
-rw-r--r-- 1 root root    40356 2009-01-01 08:00 CarroisGothicSC-Regular.ttf
-rw-r--r-- 1 root root    53256 2009-01-01 08:00 ComingSoon.ttf
-rw-r--r-- 1 root root    78984 2009-01-01 08:00 CutiveMono.ttf
-rw-r--r-- 1 root root   133636 2009-01-01 08:00 DancingScript-Regular.ttf
lrw-r--r-- 1 root root       18 2009-01-01 08:00 DroidSans-Bold.ttf -> Roboto-Regular.ttf
lrw-r--r-- 1 root root       18 2009-01-01 08:00 DroidSans.ttf -> Roboto-Regular.ttf
-rw-r--r-- 1 root root   108128 2009-01-01 08:00 DroidSansMono.ttf
-rw-r--r-- 1 root root  2779576 2009-01-01 08:00 NotoColorEmoji.ttf
-rw-r--r-- 1 root root   912352 2009-01-01 08:00 NotoColorEmojiFlags.ttf
-rw-r--r-- 1 root root   115784 2009-01-01 08:00 NotoNaskhArabic-Bold.ttf
-rw-r--r-- 1 root root   116716 2009-01-01 08:00 NotoNaskhArabic-Regular.ttf
-rw-r--r-- 1 root root   119356 2009-01-01 08:00 NotoNaskhArabicUI-Bold.ttf
-rw-r--r-- 1 root root   120800 2009-01-01 08:00 NotoNaskhArabicUI-Regular.ttf
-rw-r--r-- 1 root root    99448 2009-01-01 08:00 NotoSansAdlam-VF.ttf
-rw-r--r-- 1 root root    13852 2009-01-01 08:00 NotoSansAhom-Regular.otf
-rw-r--r-- 1 root root   134908 2009-01-01 08:00 NotoSansAnatolianHieroglyphs-Regular.otf
-rw-r--r-- 1 root root    33844 2009-01-01 08:00 NotoSansArmenian-VF.ttf
-rw-r--r-- 1 root root    11852 2009-01-01 08:00 NotoSansAvestan-Regular.ttf
-rw-r--r-- 1 root root    32084 2009-01-01 08:00 NotoSansBalinese-Regular.ttf
-rw-r--r-- 1 root root   133732 2009-01-01 08:00 NotoSansBamum-Regular.ttf
-rw-r--r-- 1 root root     6332 2009-01-01 08:00 NotoSansBassaVah-Regular.otf
-rw-r--r-- 1 root root    12808 2009-01-01 08:00 NotoSansBatak-Regular.ttf
-rw-r--r-- 1 root root   249244 2009-01-01 08:00 NotoSansBengali-VF.ttf
-rw-r--r-- 1 root root   246772 2009-01-01 08:00 NotoSansBengaliUI-VF.ttf
-rw-r--r-- 1 root root   100344 2009-01-01 08:00 NotoSansBhaiksuki-Regular.otf
-rw-r--r-- 1 root root    23652 2009-01-01 08:00 NotoSansBrahmi-Regular.ttf
-rw-r--r-- 1 root root     7844 2009-01-01 08:00 NotoSansBuginese-Regular.ttf
-rw-r--r-- 1 root root     4896 2009-01-01 08:00 NotoSansBuhid-Regular.ttf
-rw-r--r-- 1 root root 32355424 2009-01-01 08:00 NotoSansCJK-Regular.ttc
-rw-r--r-- 1 root root    62852 2009-01-01 08:00 NotoSansCanadianAboriginal-Regular.ttf
-rw-r--r-- 1 root root     6260 2009-01-01 08:00 NotoSansCarian-Regular.ttf
-rw-r--r-- 1 root root    29512 2009-01-01 08:00 NotoSansChakma-Regular.otf
-rw-r--r-- 1 root root    22272 2009-01-01 08:00 NotoSansCham-Bold.ttf
-rw-r--r-- 1 root root    22264 2009-01-01 08:00 NotoSansCham-Regular.ttf
-rw-r--r-- 1 root root    65132 2009-01-01 08:00 NotoSansCherokee-Regular.ttf
-rw-r--r-- 1 root root    21984 2009-01-01 08:00 NotoSansCoptic-Regular.ttf
-rw-r--r-- 1 root root   500380 2009-01-01 08:00 NotoSansCuneiform-Regular.ttf
-rw-r--r-- 1 root root     7980 2009-01-01 08:00 NotoSansCypriot-Regular.ttf
-rw-r--r-- 1 root root    10512 2009-01-01 08:00 NotoSansDeseret-Regular.ttf
-rw-r--r-- 1 root root   253156 2009-01-01 08:00 NotoSansDevanagari-VF.ttf
-rw-r--r-- 1 root root   247460 2009-01-01 08:00 NotoSansDevanagariUI-VF.ttf
-rw-r--r-- 1 root root   505436 2009-01-01 08:00 NotoSansEgyptianHieroglyphs-Regular.ttf
-rw-r--r-- 1 root root     8788 2009-01-01 08:00 NotoSansElbasan-Regular.otf
-rw-r--r-- 1 root root   448888 2009-01-01 08:00 NotoSansEthiopic-VF.ttf
-rw-r--r-- 1 root root    63180 2009-01-01 08:00 NotoSansGeorgian-VF.ttf
-rw-r--r-- 1 root root    15676 2009-01-01 08:00 NotoSansGlagolitic-Regular.ttf
-rw-r--r-- 1 root root     6032 2009-01-01 08:00 NotoSansGothic-Regular.ttf
-rw-r--r-- 1 root root   399684 2009-01-01 08:00 NotoSansGrantha-Regular.ttf
-rw-r--r-- 1 root root   117020 2009-01-01 08:00 NotoSansGujarati-Bold.ttf
-rw-r--r-- 1 root root   120040 2009-01-01 08:00 NotoSansGujarati-Regular.ttf
-rw-r--r-- 1 root root   114312 2009-01-01 08:00 NotoSansGujaratiUI-Bold.ttf
-rw-r--r-- 1 root root   117380 2009-01-01 08:00 NotoSansGujaratiUI-Regular.ttf
-rw-r--r-- 1 root root    32336 2009-01-01 08:00 NotoSansGunjalaGondi-Regular.otf
-rw-r--r-- 1 root root    63884 2009-01-01 08:00 NotoSansGurmukhi-VF.ttf
-rw-r--r-- 1 root root    64008 2009-01-01 08:00 NotoSansGurmukhiUI-VF.ttf
-rw-r--r-- 1 root root    13432 2009-01-01 08:00 NotoSansHanifiRohingya-Regular.otf
-rw-r--r-- 1 root root     6768 2009-01-01 08:00 NotoSansHanunoo-Regular.ttf
-rw-r--r-- 1 root root     4388 2009-01-01 08:00 NotoSansHatran-Regular.otf
-rw-r--r-- 1 root root    17540 2009-01-01 08:00 NotoSansHebrew-Bold.ttf
-rw-r--r-- 1 root root    17352 2009-01-01 08:00 NotoSansHebrew-Regular.ttf
-rw-r--r-- 1 root root     5632 2009-01-01 08:00 NotoSansImperialAramaic-Regular.ttf
-rw-r--r-- 1 root root     5668 2009-01-01 08:00 NotoSansInscriptionalPahlavi-Regular.ttf
-rw-r--r-- 1 root root     7508 2009-01-01 08:00 NotoSansInscriptionalParthian-Regular.ttf
-rw-r--r-- 1 root root    86944 2009-01-01 08:00 NotoSansJavanese-Regular.otf
-rw-r--r-- 1 root root    59452 2009-01-01 08:00 NotoSansKaithi-Regular.ttf
-rw-r--r-- 1 root root   211752 2009-01-01 08:00 NotoSansKannada-VF.ttf
-rw-r--r-- 1 root root   212632 2009-01-01 08:00 NotoSansKannadaUI-VF.ttf
[... 115 lines cut ...]
-rw-r--r-- 1 root root   249112 2009-01-01 08:00 NotoSerifKannada-VF.ttf
-rw-r--r-- 1 root root    48076 2009-01-01 08:00 NotoSerifKhmer-Bold.otf
-rw-r--r-- 1 root root    40688 2009-01-01 08:00 NotoSerifKhmer-Regular.otf
[... cut ...]
-rw-r--r-- 1 root root   305656 2009-01-01 08:00 RobotoStatic-Regular.ttf
-rw-r--r-- 1 root root   290916 2009-01-01 08:00 SourceSansPro-Bold.ttf
-rw-r--r-- 1 root root   103200 2009-01-01 08:00 SourceSansPro-BoldItalic.ttf
-rw-r--r-- 1 root root   103828 2009-01-01 08:00 SourceSansPro-Italic.ttf
-rw-r--r-- 1 root root   293516 2009-01-01 08:00 SourceSansPro-Regular.ttf
-rw-r--r-- 1 root root   291864 2009-01-01 08:00 SourceSansPro-SemiBold.ttf
-rw-r--r-- 1 root root   103556 2009-01-01 08:00 SourceSansPro-SemiBoldItalic.ttf

birdfly, 2025-06-18:
> What are the language settings?
> 
> It has few Roboto fonts (missing Roboto-Black.ttf and Roboto-Medium.ttf), but Roboto-Regular.ttf should be enough and it contains the Chinese fonts.

Chinese Simplified

birdfly, 2025-06-18:
> And did you try setting Juggluco to English in left menu->Settings->Display ?
> 
> Can you pull the Chinese fonts from your device and send them to me, so that I can try them out:
> 
> adb pull /system/fonts/NotoSansCJK-Regular.ttc
> adb pull /system/fonts/NotoSerifCJK-Regular.ttc

Just set it to “en”, now it works ,thanks for your help

closed 2025-06-19 by birdfly

j-kaltes, 2025-06-20:
It turned to be a general Android 15 issue:
https://github.com/nothings/stb/issues/509#issuecomment-2437338073
This should solve it:
https://drive.google.com/file/d/1H8roF-KrcQM_WBt-ZnbzxU5T6ZDR65zY/view