j-kaltes/Juggluco #318: Sensoren arbeiten verkürzt mit der juggluco-App

Eggi64, 2025-07-14:
Dear Ladies and Gentlemen, 
 
I have been using the Juggluco app since March 2025. I am very satisfied with the app. 
 
Unfortunately, since I used the Juggluco app, the lifespan of the sensors has basically shortened by about 2 days. The sensor is ended with sensor errors after 12 days and no longer sends values.  It can then no longer be scanned with NFC. I use a Galaxy A55 from Samsung and Libre 3 sensors.
The error occurs with sensors of different durability data. So it cannot be a mistake for a specific sensor series. 
 
The error does not occur with the sensors (also Libre 3) of my wife, who uses the Juggluco app with a Galaxy A52 from Samsung. With her, the sensors run to the displayed end date. 
 
What can I do? Due to the shortened lifespan of the sensors, the number of my sensors is no longer sufficient.
But I want to stay with Juggluco and don't switch back to the Libre 3 app. 

In the facility I will send you 2 screenshots from my smartphone.
 
Thank you. 
 
Best regards 
 
Egbert Mischke
D-02828 Görlitz

____________________________________________________________________________________________________________________________________________________________

Sehr geehrte Damen und Herren, 
 
ich nutze die juggluco-App jetzt seit März 2025. Ich bin mit der App sehr zufrieden. 
 
Leider hat sich, seit ich die juggluco-App nutze, die Lebensdauer der Sensoren grundsätzlich um etwa um 2 Tage verkürzt. Der Sensor wird nach 12 Tagen mit Sensorfehler beendet und sendet keine Werte mehr.  Er lässt sich dann auch nicht mehr mit NFC scannen. Ich nutze ein Galaxy A55 von Samsung und Libre 3-Sensoren.
Der Fehler tritt bei Sensoren unterschiedlicher Haltbarkeitsdaten auf. Es kann also kein Fehler einer bestimmten Sensor-Serie sein. 
 
Der Fehler tritt bei den Sensoren (auch Libre 3) meiner Frau, die die juggluco-App mit einem Galaxy A52 von Samsung nutzt, nicht auf. Bei ihr laufen die Sensoren bis zum angezeigten Ende-Datum. 
 
Was kann ich tun? durch die Verkürzte Lebensdauer der Sensoren, reicht die Anzahl meiner Sensoren irgendwann nicht mehr aus.
Ich möchte aber bei juggluco bleiben und nicht wieder zur Libre 3 App zurück wechseln. 

In der Anlage sende ich noch 2 Screenshots von meinem Smartphone. mit. 

Vielen Dank. 
 
Mit freundlichen Grüßen 
 
Egbert Mischke
D-02828 Görlitz

[image]
[image]

j-kaltes, 2025-07-14:
When a sensor ends with state=7 that has nothing to do with using Juggluco.
Juggluco doesn't do anything that could create that problem. Others who use Juggluco don't have it.
A few days ago the Libre 3 sensor I used ended after 14 days and 12 minutes, exactly as it should.
But ask others about it at: https://insulinclub.de
I started the sensor with Juggluco and didn't scan the sensor with Abbott's Libre 3 app, but earlier when writing the code for Libre 3 in Juggluco, I switched a lot between Juggluco and Abbott's Libre 3 apps and these sensors could also hold for 14 days, so the sensor seems to be able to handle the extra load needed to send the data to multiple apps.

Wenn ein Sensor mit dem Status=7 endet, hat das nichts mit der Verwendung von Juggluco zu tun.
Juggluco macht nichts, was dieses Problem verursachen könnte. Andere Juggluco-Nutzer haben es nicht.
Vor ein paar Tagen stoppte der Libre-3-Sensor, den ich verwendet habe, nach 14 Tagen und 12 Minuten, genau wie vorgesehen.
Aber fragen Sie andere dazu unter: https://insulinclub.de
Ich habe den Sensor mit Juggluco gestartet und ihn nicht mit Abbotts Libre-3-App gescannt. Als ich jedoch zuvor den Code für Libre 3 in Juggluco geschrieben habe, habe ich häufig zwischen Juggluco- und Abbotts Libre-3-Apps gewechselt. Diese Sensoren konnten ebenfalls 14 Tage lang halten. Der Sensor scheint also die zusätzliche Belastung bewältigen zu können, die durch das Senden der Daten an mehrere Apps entsteht.

Puderzo, 2025-08-04:
My son has had the same issues as @Eggi64 , using a **Samsung Galaxy A55**: sensor ends suddenly with "_<SensorID> stopped working. state=7_", and can not be scanned by the Libre 3 app (scanning does not pick up anything). 
[image]

We've been using Juggluco since 2022 for my son, and even though some sensors sometimes have ended prematurely for some reason or other (mostly they got loose or some such), the majority have lasted quite stably for 13 days, 23 hours and 12 minutes, plus a few seconds (the Libre2:s lasted more than 14 days, because Juggluco was able to extend the lifetime for another day). The "missing" hour from 14 days comes from the start-up period at the start of each sensor, as these times are calculated from first stream to last. This data is from the Juggluco server which we've been running for about the last two years. I have attached a parsed output of the sensors we've used and their start and end times, along with calculated lifetimes:
[sensortimes_masked.txt](https://github.com/user-attachments/files/21580642/sensortimes_masked.txt)

As you can see, since changing phones from a Samsung Galaxy S10e to a Samsung Galaxy A55 on May 24th (in the middle of a sensor, mind you, which itself went smoothly), we've had a drop of lifetime to about a bit less than 11 days, and a few dying after only a day or two. Since the change no sensor has lasted the "full" almost 14 days, even after 8 sensor changes . (NB: The last sensor in the list is still in use, even though it's reported in the file as ended today; it's a Libre 3+, and we're hoping it will last longer than the 10-11 days, but I won't know if that's the case until almost two weeks from now.)

The phone is running Android 14, because it's also running AAPS connected to an Omnipod Dash pump, which has had some issues with Android 15.

So this would strongly suggest that the phone, the **A55**, is the culprit here. I have no clue what it does differently than the S10e we used previously, but all the evidence points to that. I can see that @Ficsy (https://github.com/j-kaltes/Juggluco/discussions/288) and @ThorstenWSchmidt (https://github.com/j-kaltes/Juggluco/issues/246) also have A55:s, with similar issues of state=7, which would give further credibility to this.

ThorstenWSchmidt, 2025-08-04:
I found a solution that extends the lifetime of a Libre3+ to 15 days and about 12 minutes on my A55.

The APP Automate (https://play.google.com/store/apps/details?id=com.llamalab.automate) is able to enable and disable Bluetooth at a certain time without requiring a rooted device. I do the BT on/off during night times when I sleep. I disable BT every 56 minutes and have it enable it for 6 minutes during that time. 6 minutes is because my watch use BT as well and it requests a new blood sugar value every 5 minutes. Otherwise 2 minutes would be more than sufficient.
The connection is about 50 hours disabled per week. That is about 30% which extends the sensors lifetime for about 30-35%.

I still experiment with lower BT off times. But one try every two weeks consumes quite a time.

Back to the APP. Be warned, to have access to BT on/off you need to install another app from the vendor to have access to the legacy (Android 5) features. The vendors webpage describes the official way using Androids Debug Bridge (adb) software. I installed the downloaded APK, but of course you need to trust the vendor to not include malicious code. So never do this without caution.
I trusted the vendor and can say, it works. My sensors stay alive longer. I attach the Automate file that I am using. You need to unzip it to get a .flo file that you can load to Automate.

[AutomateFlow.zip](https://github.com/user-attachments/files/21583027/AutomateFlow.zip)

j-kaltes, 2025-08-04:
I think that it is purely accidental. 
After using Libre 2 sensors for years, I am now put on Libre 3 sensors and wear my fourth Libre3 sensor and all 3 finished sensors worked for more than 14 days and 12 minutes, exactly as expected. I did nothing unusual: let it work only with Juggluco and never with any other app and also didn’t scan the sensor with more than one exemplar of Juggluco. I only switched between phone and watch with left menu→Watch→WearOS config→”Direct sensor watch connection”.
Maybe you have placed the sensor in such a way that blood doesn’t reach it very well.
I see no way why the model of the phone would matter. A phone that constantly disconnect will give extra work, because there is an extensive exchange during the connection phase, but no phone will work so bad as Samsung Galaxy Watch5-7 and Ultra watches and they don’t seem to harm the duration of the sensor.
There is not reason why turning Bluetooth off would lengthen the duration of a sensor. Because of the above, it will also not shorten it. 
It is just magical thinking.

Ficsy, 2025-08-04:
I can only confirm the same. Since I’ve been using an A55 phone, the sensor stops working after about 10 days. Before that, I used an older Samsung phone and had no issues. I’m almost certain that the Bluetooth connection is draining it.