j-kaltes/Juggluco #322: Bluetooth connection to sensor not working anymore

kdupke, 2025-07-25:
phone: Juggluco 9.7.7-phone-li3-si-x-g
watch: 9.7.6-wear-li3-si-x-nog 731

Hi,

I have a running sensor. Since today the bluetooth connection between my mobile and the sensor is broken.
However, the connection from sensor to watch to mobile phone is working.

So for me it looks like sensor is working and bluetooth is working on all devices too. As I do not have wifi activated.

When I open juggluco it says no connection. After some time, around 1 minute, juggluco closes (while display is still on).

Of course I have done nothing special :-) 

Any ideas? 

Best regards,
kai

j-kaltes, 2025-07-26:
Install a logging version of Juggluco for the phone (https://www.juggluco.nl/Juggluco/download.html#bugreport), let it try to connect with the sensor until it closes automatically and e-mail me trace.log and logcat.txt.

Why don't you use the Google Play version on your watch?

kdupke, 2025-07-26:
Thank you for prompt feedback.
Will have the log soon.
The phone and watch are not connected to google. The phone uses aurora for APKs, the watch can't use aurora, so I install the packages manually via the 'wear installer'.

kdupke, 2025-08-01:
Sorry, took a while.
Tried with 10.0.2-phone-li3-si-x-nog from uptodown , but no Settings -> Logging there.
What is the name of a version I can do logging with? Or is it missed in the described google play issues?
Again sorry, if I misinterpret your directions

j-kaltes, 2025-08-01:
The release version doesn't log, you need a special logging version: https://drive.google.com/file/d/1SinzSL7THorHx6x9CM5n63ugQmFyj5cQ/view

kdupke, 2025-08-01:
That worked well. Email sent.

j-kaltes, 2025-08-01:
Yes, Juggluco crashes. 
What phone and what custom system do you exactly use?
Do you have root, Magisk and/or GAPPS (including Google Play services)?

j-kaltes, 2025-08-01:
Magisk doesn't make sense without root. Do you have root and hide it with Magisk?
With certain magisk modules, you can change the phone information. You can make it look like it is another phone, but I don't think that will matter.

j-kaltes, 2025-08-01:
Could you not hide root for Juggluco, run Juggluco and e-mail me trace.log and logcat.txt?

kdupke, 2025-08-01:
Does the trick - it works now.

Means: Running juggluco with Magisk but not letting Magisk hide it for Juggluco breaks the Bluetooth connection and let Juggluco crash. Adding Juggluco to the list of apps to hide root/Magisk for it, it works without crash.

I thought that was the reason about Magisk that you felt root(ed) would cause the trouble.
I did nothing fancy like running strace or debug or such sort.

The logs I sent to you are with *not* hiding root.

As I do not get crashes *with* hiding root I am not sure I can create useful logs at al?

I also uninstalled Magisk completely (which uninstalls the app and disables itself from the kernel (and reboots) of course.
Then Jugglucco works as well. No crashes. 

Switching to a magisk patched kernel, but not installing the Magisk app, again let it crash. Logs are in your email.
Installing Magisk 29.0 (not enabling any options). 
Juggluco crashes. 
That is expected as this is where I started this issue.

Hiding Magisk for Juggluco. 
Everything works.

Not sure what I can do to help to find what causes this, or if I shall file an issue with magisk then?

j-kaltes, 2025-08-17:
Maybe you have installed something that becomes active when root is not hidden.

The logs I received earlier showed that root was hidden, so something else has to be the case. 
On my phone Juggluco doesn't crash when root can be detected.