j-kaltes/Juggluco #323: Feature Request: Configurable Weather Data Mapping for Broader Gadgetbridge Compatibility**

SLi95c, 2025-07-25:
**Summary:**
Currently, only a limited number of Mi Band models are supported by the Gadgetbridge fake weather data solution in Juggluco. No other smartwatch types are currently compatible with this weather-based Glucose data transmission method. Since each watch model handles weather data differently, creating watchfaces for other popular models such as the Amazfit BIP or GTR 2 is not feasible under the current setup.

**Why this is important:**
The weather data interface offers the most energy-efficient method for continuous Glucose monitoring on wearables. Compared to frequent data uploads (like those used by Watchdrip), this approach is significantly more battery-friendly—an essential factor for smartwatches.

**Proposed solution:**
If users could configure how specific weather fields (e.g., temperature, humidity, condition) are mapped to different Glucose values, the solution would become theoretically compatible with a wide range of Amazfit and Mi Band devices. It would also allow developers to build watchfaces tailored to each device’s weather data parsing behavior.

**Additional Insight:**
Discrepancies in displayed glucose values seem to stem from how Juggluco currently uses the *current temperature* (T\_current) to encode changes. On devices like the Amazfit BIP, there's internal logic that replaces the max temperature (T\_max) with T\_current if T\_current is higher (and possibly the same for T\_min if it's lower).

This causes issues when glucose levels change quickly. For example, Juggluco might encode a value as 160, but the watch ends up showing 320 because T\_current = 32 overrides T\_max = 16 (with T\_min = 0).

To avoid such behavior, T\_current must always remain between T\_min and T\_max. This kind of device-specific behavior reinforces the need for **configurable weather data mapping**, so the encoding can be tailored per device.

**Benefits:**

* Greatly expanded compatibility with many popular smartwatch models (e.g., Amazfit BIP, GTR2).
* Enables custom watchface creation for various devices using Juggluco.
* Makes Juggluco the most flexible and powerful solution for Glucose monitoring on wearables.

This enhancement would unlock powerful new possibilities for users and developers alike, making Juggluco a comprehensive and adaptable solution for the diabetes community.

j-kaltes, 2025-07-27:
I see nothing in making everything configurable. That makes everything needlessly difficult. Every user needs to set all kind of variables, and every maker of a watch face needs to decide which variable to use for what. 
Everything is much easier when one assignment is used so that the other side knows what to do.
If the current assignment gives problems, another has to be selected.

lgs11, 2026-11-03:
I have a Pinetime watch from Pine64, running Pinetime 1.16. It can display the current temperature and weather symbol on the default watch face. It would be great if there could be an option to report the glucose value only as the current temp.

I'm in Sweden, so use mmol/L and Celsius. Maybe three digit temperatures for mg/dL is not supported, though. Personally, I think rounding to an integer would be ok if necessary.

Here are images of the value 7.1 on the main face and on the weather app.

[image]
[image]

j-kaltes, 2026-11-04:
ChatGPT has written a glucose on watch via Gadgetbridge tutorial:
https://github.com/j-kaltes/Juggluco/wiki/Gatgetbridge