j-kaltes/Juggluco #326: FR: Change http port in Juggluco cmd line

Stan-Di, 2025-07-28:
Hello @j-kaltes 

Please add the ability to select the http port in Juggluco cmd line. Now I can only configure the https port, http is always only 17580. I have an nginx server with SSL certificates and I do not need an additional https web server from Juggluco.

j-kaltes, 2025-07-28:
In no time you can ask for a free SSL certificate with certbot:
```sh
certbot certonly --standalone --preferred-challenges http -d myhostname
```
(If you forward port 80 from you modem and run no other http server).
See: https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html

Stan-Di, 2025-07-28:
Yes, I know how to do it. And I use it now. But I had to switch to the static linked version. And the -o port option only works for SSL port. I would like to run several Juggluco cmdlines like now, but each with its own http port, not 17580.

j-kaltes, 2025-07-28:
Why not the dynamically linked or a self-compiled version?

Stan-Di, 2025-07-29:
> In Juggluco server 10.0.1 you can change the http port with -W port
> 
> https://www.juggluco.nl/Juggluco/cmdline/index.html

Hello! I try option -W port but -l say me "eXport web server turned on (port 17580)"

[image]

Stan-Di, 2025-07-29:
In Juggluco 10 I find trouble

I tested 1 instance and in /x/report  begin load cpu and memory...

[image]

I stopped it and load decreased

Full log:

> Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346682 before showscans
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346682 before showsnums catnr=0
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346682 showNums number= 0175377144
0 346682 showNums number= 01753771440 346682 end display no value
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 displaycurve histlen=0 docl
amp=0 starttime=1752699598
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before getranges
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 Before numdatas[i]->getInRa
nge(1752699598,1752785998)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 Before getextremes
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 0 before extremenums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 extreamenums(6000,0)->(6000
,0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 1 before extremenums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 extreamenums(6000,0)->(6000
,0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before gettrans
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 getdisplaytime xscale=0.017
778 3600 1752699600 1752786000
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showsaverange
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showNums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 duration=86400
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 timdis=1071842 duration=864
00 grens=3
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 timdis=1071842 larger than zero
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346684 displaycurve histlen=0 doclamp=0 starttime=1753736398
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346684 before getranges
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346684 Before numdatas[i]->getInRange(1753736398,1753822798)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 displaycurve histlen=0 doclamp=0 starttime=1752181198
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before getranges
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 Before numdatas[i]->getInRange(1752181198,1752267598)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 Before getextremes
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 0 before extremenums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 extreamenums(6000,0)->(6000,0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 1 before extremenums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 extreamenums(6000,0)->(6000,0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before gettrans
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 getdisplaytime xscale=0.017778 3600 1752181200 1752267600
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showsaverange
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showNums
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 duration=86400
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 timdis=1590242 duration=86400 grens=3
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 timdis=1590242 larger than zero
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 displaytime Fri 11-07-2025
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 glucoselines: unit=90.000000 unit2=180.000000 step=180 (1) startl=0 (0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showhistories
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showcalibrated
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showstream
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showscans
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 before showsnums catnr=0
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346665 showNums number= 01753771440 346665 showNums number= 01753771440 346665 end display no value
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 displaytime Thu 17-07-2025
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 glucoselines: unit=90.000000 unit2=180.000000 step=180 (1) startl=0 (0)
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showhistories
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showcalibrated
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showstream
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showscans
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 before showsnums catnr=0
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346677 showNums number= 01753771440 346677 showNums number= 01753771440 346677 end display no value
Jul 29 09:44:01 x4 jug.sh[344659]: 1753771440 346684 Before getextremes
[... 922 lines cut ...]
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: ack send
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcom
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: got tag
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: datlen=8
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: voor recv
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: voor ascon_aead128a_decrypt_update
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: ascon_aead128a_decrypt_final valid
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: interpret len=8
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: 67 com=srender (8) 0
[... cut ...]
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 344661 0 before lock
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcommands: ack send
Jul 29 09:45:08 x4 jug.sh[344659]: 1753771508 347101 getcom
Jul 29 09:45:31 x4 systemd[1]: Stopping jug10@8000.service - Juggluco 10 Service 8000...
Jul 29 09:45:31 x4 systemd[1]: jug10@8000.service: Deactivated successfully.
Jul 29 09:45:31 x4 systemd[1]: Stopped jug10@8000.service - Juggluco 10 Service 8000.
Jul 29 09:45:31 x4 systemd[1]: jug10@8000.service: Consumed 8min 37.412s CPU time, 194.1M memory peak.

j-kaltes, 2025-07-29:
You don't want to disallow summarygraph? That is the only one that is so terribly slow.

Stan-Di, 2025-07-30:
I guess I didn't understand yesterday what you meant by "disallow summarygraph".

Regarding JavaScript - Chart.js is completely free and opensourse - https://github.com/chartjs/Chart.js. I found examples at https://www.w3schools.com/ai/ai_chartjs.asp

I played around a bit and created this html page https://www.w3schools.com/js/tryit.asp?filename=tryai_chartjs_bars_colors_more :

```
<!DOCTYPE html>
<html>
<script src="https://cdnjs.cloudflare.com/ajax/libs/Chart.js/2.5.0/Chart.min.js"></script>
<body>

<canvas id="myChart" style="width:100%;max-width:600px"></canvas>

<script>
var xValues = ["2025-07-30T05:37:00.000Z", "2025-07-30T05:38:00.000Z", "2025-07-30T05:39:00.000Z", "2025-07-30T05:40:00.000Z", "2025-07-30T05:41:00.000Z"];
var yValues = [172, 168, 175, 167, 180];

new Chart("myChart", {
  type: "line",
  data: {
    labels: xValues,
    datasets: [{
      data: yValues
    }]
  },
  options: {
    legend: {display: false},
    title: {
      display: true,
      text: "Juggluco data"
    }
  }
});
</script>

</body>
</html>
```

This is how it looks in the browser

[image]

This data can be taken from json to create an html page

```
var xValues = ["2025-07-30T05:37:00.000Z", "2025-07-30T05:38:00.000Z", "2025-07-30T05:39:00.000Z", "2025-07-30T05:40:00.000Z", "2025-07-30T05:41:00.000Z"];
var yValues = [172, 168, 175, 167, 180];
```