j-kaltes/Juggluco #332: Bug: 10.0.2. Not working menu "sensor"

Stan-Di, 2025-08-01:
In Juggluco 10.0.2 "sensor" button stopped working. I made a video - https://disk.yandex.ru/i/9dqWvh3q6m0bYg

I click there and nothing happens.

j-kaltes, 2025-08-01:
That can happen, but only when you don't have an active sensor.
What did left menu->Sensor show before it stopped showing anything?

j-kaltes, 2025-08-01:
Does this version still have this issue: https://drive.google.com/file/d/1IQgm6PpjMeCeomCzIhELgMtglsPP-_z1/view

Stan-Di, 2025-08-03:
> Does this version still have this issue: https://drive.google.com/file/d/1IQgm6PpjMeCeomCzIhELgMtglsPP-_z1/view

Yes, this version works

[image]

I use my Juggluco as a receiver for my child's readings. It's convenient for me when his phone dies, that I can switch Bluetooth on mine and get readings.