j-kaltes/Juggluco #338: Two Watches?

greenie95125, 2025-08-20:
I now have two Samsung Galaxy Watch 6 Classics. I got the second one (diff color) as a spare, but I decided that I may want to swap them out on occasion. I have a Pixel 7 Pro, and Juggluco is on all 3 devices. For the last couple of years, I've been making the sensor connection (3+ now) with my watch and send the values to my phone. I love the convenience of not needing my phone within 30 feet to see my values. Kudos to the developer for making this much easier to do. I know both watches can do this because when I set up the "new" watch, I was able to make the watch sensor connection with it. However, now I can't get it to switch back.

In the Wear app, I've changed the connection to the silver(old) watch, I've turned off BT on the black watch, I checked jug on the phone and switched the watch setting to the silver watch, and rebooted everything. The silver watch just says "no value." What am I missing? Is there a relatively simple method to do this? I know that I can revert to the old way and have the phone connect with the sensor, then send values to the watch. That would be easier (I think) to switch between watches, but I would much prefer to get the Watch - Sensor connection going again when I switch. Thanks for any insight the community may have for me.

j-kaltes, 2025-08-20:
Juggluco itself has no problem with connecting with multiple watches at the same time, but Samsung Galaxy Wearable app has. When you use watches that make use of different companion apps on the phone, both watches receive values from Juggluco at the same time. 
When I tried two watches that both make use of the Samsung Galaxy Wearable app, I experienced all kinds of problems. First problem was that when I paired the new watch with the Wearable app, it forgot the old watch. But after pairing the old watch again, I manage to have them both represented in the Wearable app. But the Wearable app doesn't connect to both of them at the same time, you have to select one of them and only the selected one will communicate via Bluetooth.

[image]

[image]

[image]

[image]

greenie95125, 2025-08-20:
Good morning Jaap -

I figured the wearable app was the roadblock for having both active concurrently. Because of that, I was just looking for a seamless way to switch from one watch to another. I'm still experimenting. The only way (so far) I can find to make it work (so far) is to first make the phone connect to the sensor, wait until I see a value, then make the switch in the wearable app. Once I see a value on the watch, then go back to the phone and check the "Direct sensor-watch connection" box. I'm not even 100% sure if those were the exact steps. I'll try again tonight after work and write stuff down this time. Currently, I do have the alternate watch connected to the sensor and sending to the phone which was my goal yesterday. I'd just like to know if there is a simple way that should work

As always, that you for your help, and for developing such a fantastic app for all of us

Select.png (view on web) [https://github.com/user-attachments/assets/f9ba7e2c-1e42-4a3f-80f5-77a5abda2752] connecting.png (view on web) [https://github.com/user-attachments/assets/fb9c918d-90c6-4e6c-8368-22e4c84fce7c] Watch7Disconnects.png (view on web) [https://github.com/user-attachments/assets/0cf1afeb-e523-43c1-b05c-dcd8a93aa32b] twoWatches.png (view on web) [https://github.com/user-attachments/assets/69e6704c-da7f-4a9e-b8f5-013b5ace5be7]
[842e7520-1f8b-4d35-81b5-0d2a7596b440]