j-kaltes/Juggluco #340: Data discrepancies between Juggluco and the Sibionics official app

DengBuyun1, 2025-08-24:
I used Sibionics' "Chuguomen" to collect about 10 days of data, but the CGM unexpectedly detached. A few days later I used Juggluco to retrieve data from that CGM and compared it with the data exported from the Sibionics official app; not only were the glucose values different, but the timestamps also differed. Could this be caused by a CGM malfunction? (The CGM I used is from Chinese Sibionics.)

[111.xlsx](https://github.com/user-attachments/files/21956873/111.xlsx)

DengBuyun1, 2025-08-25:
First of all, I’m glad to receive your reply, and I must admit I did make some mistakes.

“Chuguomen” actually refers to the “CGM”.
By “异常,” I mean situations where the Sibionics CGM either falls off before the required number of days and the sensor detects erroneous data—in such cases, the CGM will report it as “异常.”
The file I attached previously also contained errors, and I will resend the correct one.
Additionally, today I reinstalled version 10.0.2 and connected to the same CGM. The data displayed in the app is shown in the attached image, where you can see that the app shows the start date as August 1st. I’m not sure if this is due to an abnormality with the Sibionics CGM.

[image]
[image]
[image]

[蓝牙2025-08-09-20_23_15.tsv](https://github.com/user-attachments/files/21963798/2025-08-09-20_23_15.tsv)

j-kaltes, 2025-08-25:
The new files still put the wrong values after each other. Look at the first data line:
``` tsv
U6DY670XAH2 1   1754045503  2025-08-01-18:51:43 8   5   5.8 6.5 NOT_DETERMINED      2025/07/12 10:21    4.7
```
You have put   2025/07/12 10:21 after  2025-08-01-18:51:43.
You can better keep them in separate files.

DengBuyun1, 2025-08-25:
I would like to clarify that the following data, "2025/07/12 10:21 4.7," is derived from the official Sibionics application.

j-kaltes, 2025-08-25:
Yes, why do you append "2025/07/12 10:21" after "2025-08-01-18:51:43". There is no relation between these data.

DengBuyun1, 2025-08-25:
> The data on the left is what I received using Juggluco to collect CGM data, while the data on the right was exported from the official Sibionics app. Previously (on July 12th), I activated the CGM using the official app and it ran for about 10 days. The CGM then fell off, which caused some abnormalities in data monitoring. Today, I connected this CGM to Juggluco, and the app received the data shown above. I believe the amount of data received by Juggluco is correct (both the Sibionics app and Juggluco have about 10 days of data). However, the issue is that the sequence of the data exported from the official app is different from what Juggluco received. I am not sure if this discrepancy is due to the monitoring abnormalities caused by the CGM falling off

Because I activated the CGM using the official app on July 12th, but later the CGM fell off and the app showed that the CGM was abnormal. Today, I used Juggluco to connect to this CGM and retrieve its stored historical data. However, I found that the start date of the received data was August 1st. In fact, this CGM had already stopped monitoring glucose data on July 21st (it was showing abnormal status), but the data exported by Juggluco started from August 1st (as shown in the app screenshot above).